arduino-libraries/AudioFrequencyMeter #10: Library Not Working

JamieStivala, 2018-03-10:
This library is not even compiling

per1234, 2018-03-10:
When you encounter an error in the Arduino IDE you'll see a button on the right side of the orange bar "Copy error messages". Click that button. Paste the error in a message here using code tags (**<>** button on the toolbar).

JamieStivala, 2018-03-10:
<

WARNING: library AudioFrequencyMeter claims to run on (samd) architecture(s) and may be incompatible with your current board which runs on (avr) architecture(s).
/Users/jamiestivala/Documents/Arduino/libraries/AudioFrequencyMeter/src/AudioFrequencyMeter.cpp: In member function 'void AudioFrequencyMeter::ADCconfigure()':
/Users/jamiestivala/Documents/Arduino/libraries/AudioFrequencyMeter/src/AudioFrequencyMeter.cpp:163:6: error: base operand of '->' is not a pointer
   ADC->CTRLB.bit.RESSEL = ADC_CTRLB_RESSEL_8BIT_Val;
      ^
/Users/jamiestivala/Documents/Arduino/libraries/AudioFrequencyMeter/src/AudioFrequencyMeter.cpp:163:27: error: 'ADC_CTRLB_RESSEL_8BIT_Val' was not declared in this scope
   ADC->CTRLB.bit.RESSEL = ADC_CTRLB_RESSEL_8BIT_Val;
                           ^
/Users/jamiestivala/Documents/Arduino/libraries/AudioFrequencyMeter/src/AudioFrequencyMeter.cpp:167:6: error: base operand of '->' is not a pointer
   ADC->CTRLB.bit.PRESCALER = ADC_CTRLB_PRESCALER_DIV8_Val;     // Divide Clock by 8 -> ~100kHz
      ^
/Users/jamiestivala/Documents/Arduino/libraries/AudioFrequencyMeter/src/AudioFrequencyMeter.cpp:167:30: error: 'ADC_CTRLB_PRESCALER_DIV8_Val' was not declared in this scope
   ADC->CTRLB.bit.PRESCALER = ADC_CTRLB_PRESCALER_DIV8_Val;     // Divide Clock by 8 -> ~100kHz
                              ^
/Users/jamiestivala/Documents/Arduino/libraries/AudioFrequencyMeter/src/AudioFrequencyMeter.cpp:175:6: error: base operand of '->' is not a pointer
   ADC->SAMPCTRL.reg = 0x1F;                                    // Set max Sampling Time Length
      ^
/Users/jamiestivala/Documents/Arduino/libraries/AudioFrequencyMeter/src/AudioFrequencyMeter.cpp: In function 'bool ADCisSyncing()':
/Users/jamiestivala/Documents/Arduino/libraries/AudioFrequencyMeter/src/AudioFrequencyMeter.cpp:184:14: error: base operand of '->' is not a pointer
   return (ADC->STATUS.bit.SYNCBUSY);
              ^
/Users/jamiestivala/Documents/Arduino/libraries/AudioFrequencyMeter/src/AudioFrequencyMeter.cpp: In member function 'void AudioFrequencyMeter::ADCdisable()':
/Users/jamiestivala/Documents/Arduino/libraries/AudioFrequencyMeter/src/AudioFrequencyMeter.cpp:189:6: error: base operand of '->' is not a pointer
   ADC->CTRLA.bit.ENABLE = 0x00;                               // Disable ADC
      ^
/Users/jamiestivala/Documents/Arduino/libraries/AudioFrequencyMeter/src/AudioFrequencyMeter.cpp: In member function 'void AudioFrequencyMeter::ADCenable()':
/Users/jamiestivala/Documents/Arduino/libraries/AudioFrequencyMeter/src/AudioFrequencyMeter.cpp:196:6: error: base operand of '->' is not a pointer
   ADC->CTRLA.bit.ENABLE = 0x01;                               // Enable ADC
      ^
/Users/jamiestivala/Documents/Arduino/libraries/AudioFrequencyMeter/src/AudioFrequencyMeter.cpp: In member function 'void AudioFrequencyMeter::ADCsetMux()':
/Users/jamiestivala/Documents/Arduino/libraries/AudioFrequencyMeter/src/AudioFrequencyMeter.cpp:208:28: error: 'g_APinDescription' was not declared in this scope
   pinPeripheral(samplePin, g_APinDescription[samplePin].ulPinType);
                            ^
/Users/jamiestivala/Documents/Arduino/libraries/AudioFrequencyMeter/src/AudioFrequencyMeter.cpp:208:66: error: 'pinPeripheral' was not declared in this scope
   pinPeripheral(samplePin, g_APinDescription[samplePin].ulPinType);
                                                                  ^
/Users/jamiestivala/Documents/Arduino/libraries/AudioFrequencyMeter/src/AudioFrequencyMeter.cpp:212:6: error: base operand of '->' is not a pointer
   ADC->INPUTCTRL.bit.MUXPOS = g_APinDescription[samplePin].ulADCChannelNumber; // Selection for the positive ADC input
      ^
/Users/jamiestivala/Documents/Arduino/libraries/AudioFrequencyMeter/src/AudioFrequencyMeter.cpp: In member function 'void AudioFrequencyMeter::tcConfigure()':
/Users/jamiestivala/Documents/Arduino/libraries/AudioFrequencyMeter/src/AudioFrequencyMeter.cpp:218:3: error: 'GCLK' was not declared in this scope
   GCLK->CLKCTRL.reg = (uint16_t) (GCLK_CLKCTRL_CLKEN | GCLK_CLKCTRL_GEN_GCLK0 | GCLK_CLKCTRL_ID(GCM_TC4_TC5)) ;
   ^
/Users/jamiestivala/Documents/Arduino/libraries/AudioFrequencyMeter/src/AudioFrequencyMeter.cpp:218:35: error: 'GCLK_CLKCTRL_CLKEN' was not declared in this scope
   GCLK->CLKCTRL.reg = (uint16_t) (GCLK_CLKCTRL_CLKEN | GCLK_CLKCTRL_GEN_GCLK0 | GCLK_CLKCTRL_ID(GCM_TC4_TC5)) ;
                                   ^
/Users/jamiestivala/Documents/Arduino/libraries/AudioFrequencyMeter/src/AudioFrequencyMeter.cpp:218:56: error: 'GCLK_CLKCTRL_GEN_GCLK0' was not declared in this scope
   GCLK->CLKCTRL.reg = (uint16_t) (GCLK_CLKCTRL_CLKEN | GCLK_CLKCTRL_GEN_GCLK0 | GCLK_CLKCTRL_ID(GCM_TC4_TC5)) ;
                                                        ^
/Users/jamiestivala/Documents/Arduino/libraries/AudioFrequencyMeter/src/AudioFrequencyMeter.cpp:218:97: error: 'GCM_TC4_TC5' was not declared in this scope
   GCLK->CLKCTRL.reg = (uint16_t) (GCLK_CLKCTRL_CLKEN | GCLK_CLKCTRL_GEN_GCLK0 | GCLK_CLKCTRL_ID(GCM_TC4_TC5)) ;
                                                                                                 ^
/Users/jamiestivala/Documents/Arduino/libraries/AudioFrequencyMeter/src/AudioFrequencyMeter.cpp:218:108: error: 'GCLK_CLKCTRL_ID' was not declared in this scope
   GCLK->CLKCTRL.reg = (uint16_t) (GCLK_CLKCTRL_CLKEN | GCLK_CLKCTRL_GEN_GCLK0 | GCLK_CLKCTRL_ID(GCM_TC4_TC5)) ;
[... cut ...]
/Users/jamiestivala/Documents/Arduino/libraries/AudioFrequencyMeter/src/AudioFrequencyMeter.cpp:293:6: error: base operand of '->' is not a pointer
   ADC->SWTRIG.bit.START = 0;
      ^
/Users/jamiestivala/Documents/Arduino/libraries/AudioFrequencyMeter/src/AudioFrequencyMeter.cpp: In function 'void TC5_Handler()':
/Users/jamiestivala/Documents/Arduino/libraries/AudioFrequencyMeter/src/AudioFrequencyMeter.cpp:376:3: error: 'TC5' was not declared in this scope
   TC5->COUNT16.INTFLAG.bit.MC0 = 1;     // Clear interrupt
   ^
exit status 1
Error compiling for board Arduino/Genuino Uno.
>

per1234, 2018-03-10:
Here's your clue:
>`WARNING: library AudioFrequencyMeter claims to run on (samd) architecture(s) and may be incompatible with your current board which runs on (avr) architecture(s).`

From https://github.com/arduino-libraries/AudioFrequencyMeter/blob/master/README.adoc:
>Allows the **Arduino Zero and MKR1000** to sample a generic input audio signal and get the fundamental pitch.

The library is written for SAMD architecture but you're compiling for the Uno, which is AVR architecture. If you actually have a SAMD board then you need to select the correct board from the Arduino IDE's **Tools > Board** menu. If you don't own a SAMD board then you're out of luck, you can't use this library.

JamieStivala, 2018-03-10:
Bummer, okay

closed 2018-03-10 by JamieStivala

JamieStivala, 2018-03-12:
May I ask, what makes this library only work on SAMD architecture? Is there any specific instruction? 

Thanks

facchinm, 2018-03-12:
The library needs to run at very high (and deterministic) ADC sampling rates so right now it uses low level access to the SAMD peripherals. Of course, there is nothing preventing the library to be ported to other platforms with an ADC and a programmable timer, so we'll be glad to accept pull requests targeting other platforms :smile:

PaulStoffregen, 2018-03-12:
Would you accept a pull request for a 3rd party boards?

PaulStoffregen, 2018-03-12:
FWIW, I haven't ported this one to Teensy, and I didn't submit pull for AudioZero (which I did port) on the assumption you probably wouldn't want to merge support for non-Arduino boards.

facchinm, 2018-03-12:
Adding specific code for third party boards is a bit cumbersome since we'd need to maintain the library on platforms we don't know well (and we are not under control of the core development).IMO we could setup some sort of "support agreement" so any issue/PR related with a 3rd party board should be handled by the submitter.  @cmaglie what do you think about that?

PaulStoffregen, 2018-03-12:
If I were to port this one, my pull request would look like 2 commits.  The first would create inline functions (or maybe preprocessor defines) for the many lines scattered throughout the code doing hardware access, so all the arch specific stuff can be localized to one area.  The second would add ifdef checks on the arch or cpu, to give a separate set of defines for different hardware.

Most of the long established Arduino libraries have this approach... because I'm the one of ported them all from their original Uno-only code.

JoshBacigalupi, 2018-09-17:
Hi,
I purchased an ItsyBitsy M4 Express so I could use this library, as it has a SAMD chip.  Nevertheless, I received a similar error:

C:\Users\Admin\Documents\Arduino\libraries\AudioFrequencyMeter-master\src\AudioFrequencyMeter.cpp:376:3: error: 'TC5' was not declared in this scope

   TC5->COUNT16.INTFLAG.bit.MC0 = 1;     // Clear interrupt
   ^
    exit status 1
    Error compiling for board Adafruit ItsyBitsy M4 (SAMD51).

As this is a third party board, does this library need to be ported for this specific chip?  If not this one, are there any frequency detection libraries that work 'as is' for the ATSAMD51?

Thanks alot.